jbouwh/omnikdatalogger #2: Omnik Datalogger OSError: [Errno 99] Address not available

florisdegroot, 2021-06-17:
I am struggling with the Omnik Data logger Automation in HACS.
I am running Home Assistant running on a RPI and I have all software (HA-Core, Supervisor, HACS etc) updated to the latest releases.
I have:
- installed Appdaemon4
- updated /config/appdaemon/appdaemon.yaml (file appended as .txt)
- update  /config/appdaemon/apps/apps.yaml (file appended as .txt)
- installed the integration DSMR with an Omnik v2.2. which is running and providing 5 sensors

But I have still an error in the Appdaemon4, see the last part of the log in appended log file
[appdeamon log.log](https://github.com/jbouwh/omnikdatalogger/files/6673313/appdeamon.log.log)
[appdaemon.txt](https://github.com/jbouwh/omnikdatalogger/files/6673255/appdaemon.txt)
[apps.txt](https://github.com/jbouwh/omnikdatalogger/files/6673266/apps.txt)

Do you need more info?
Can you advise me what I am missing or doing wrong.?

Kind regards,

jbouwh, 2021-06-17:
It seems mqtt cannot connect. Create a username and password and add them to the mqtt plugin config. Does the app run without output plugins configured?

Did jou notice the Dutch wiki the website?

florisdegroot, 2021-06-18:
Thanks for the quick reply. As you could see in the appdeamon.txt (yaml) file and the logfile I have  used a mqtt username etc but I will make a new MQTT username and password and will retry. I will be back this evening.

jbouwh, 2021-06-18:
If you are using hassapi as input client, you will need to have a omnikdataloggerproxy setup to retrieve the updates using the binary sensor it creates. This is what omnikdatlogger is listening to.

florisdegroot, 2021-06-18:
1st I go to read and follow the instructions of the WIKI Dutch pages.

It looks that these are much more clear to me. Less options etc.

If I thereafter get again errors I will open again an issue.

Thanks for he support so far.

jbouwh, 2021-06-18:
v1.5.7 fixes crashing with an invalid MQTT config.
For more information about the config see:

https://github.com/jbouwh/omnikdatalogger/blob/main/info.md
https://github.com/jbouwh/omnikdatalogger#readme

or the website for a Dutch Wiki:
https://jbsoft.nl/site/dutch-only/

closed 2021-06-18 by jbouwh