jbouwh/omnikdatalogger #3: Make DSMR entities compliant with new Home Assistant "Energy" module

serrj-sv, 2021-08-05:
Latest Home Assistant  introsuced new "Energy Grid" module. But DSMR entities produced by "omnikdatalogger" does not apper in the list of "Consumed Energy" entities list

[image]

Also couple of suggestions:
1. disable polling omnik inverter in tcpclient mode from dusk to dawn to not spam the log
2. keep updating "combined DSMR" entities event if there is no updates coming from Inverter.

Anyway, thanks for amazing tool, I really enjoy it!!

jbouwh, 2021-08-05:
Did you install release 1.6.4?
There were some fixes.

serrj-sv, 2021-08-05:
yep. latest version installed

serrj-sv, 2021-08-05:
this is strange, because in dev. tools I can see all necessery attributes (device_class: energy and state_class: measurement)

jbouwh, 2021-08-05:
I've got it working myself

serrj-sv, 2021-08-05:
Oh.. indeed, it looks like I was running  old version of omnikdatalogge. Just received an update notification in HACS!

serrj-sv, 2021-08-05:
everything is OK now, closing

closed 2021-08-05 by serrj-sv