jbouwh/omnikdatalogger #13: Can't add omnik in HACS and the new version of HACS does not have an automation menu.

bright72, 2021-09-17:


jbouwh, 2021-09-17:
I am using HACS 1.15.2 and all works as designed. May be you need to restart or refresh your browser. Further, make sure you have installed the AppDaemon Addon. Without it you will not be able to run AppDaemon or NetDeamon.

jbouwh, 2021-09-17:
Closing here. Please let me know if this is not helping.

closed 2021-09-17 by jbouwh

bright72, 2021-09-17:
[image]

jbouwh, 2021-09-17:
You do not need to add it as a custom integration. It is an automation.

bright72, 2021-09-18:
I have already configured and install AppDaemon 4. Why search not found.

[image]

[image]

[image]

jbouwh, 2021-09-18:
Did you install AppDaemon4 before you installed HACS?
https://hacs.xyz/docs/categories/appdaemon_apps/

bright72, 2021-09-18:
Thank you it helps me so much.