jbouwh/omnikdatalogger #25: Can't read data from logger_sn 644xxxxxx/inverter_sn AALNxxxx

georgeavd, 2021-11-02:
Hi Jan,

For years I'm a using modified Python-script to read the actual data from <ip>/js/status.js
I recently started playing around with your omnikdatalogger because I would like to use the tcpclient setup for retrieving data from my 3 inverters.

After playing around with the config.yaml file I can now read data from the two inverters for which the inverter_sn is starting with NLxxx, and the logger_sn starting with 602xxxxxx and 634xxxxxx.
But the one inverter with an inverter_sn starting with AALNxxxx I get a 'timed out' in the logging. This inverter is running with logger_sn starting with 644xxxxxx
Can it be that the omnikdatalogger is testing if the inverter_sn is starting with NLxxx, or is this an issue related to the logger_sn?

Regards George

jbouwh, 2021-11-02:
I think the method using /js/status.js is variant that is not supported by omnikdatalogger yet. My own inverter does not support it, so it is difficult to debug that way. The `tcpclient` plugin uses a different port (8844) and proprietry protocol. It uses the code of Wouter van de Zwan. Also this client is not supported with my inverter. It is known that tcpclient only has been proven to work with certain serial numbers.

To use /js/status.js  I would need to implement a new client. But I need the content of status.js to reverse enginere that.

Alternatives with omnikdatalogger are intercepting the logger traffic with the localproxy client (the method I am using). This method has some disadvantages:
- You need to reroute traffic and do NAT to simuate the Solarman logger (works on port 10004).
- You data will no longer be send to the Solarman services (omnikdataloggerproxy supplies a work-a-round).

An other alternative is the new solarmanpv client, but it is not sure how long that method can be used and how stable the API will be, but it works.

To the current code base stictly this is not an issue with omnikdatalogger.

Feel free to collaborate to get a client based on /js/status.js implemented. I can give it a try if you can supply some raw content of this status.js using wget.

georgeavd, 2021-11-02:
If a `tcpclient` connection is not an option for this inverter then it would indeed be nice to have the `<ip>/js/status.js` functionality embedded in your omnikdatalogger.
I will provide the variants (yes lucky me, I have three Omnik inverters which all provide data in a different way) of the status.js accompanied with the screenshots of the webinterface

In status-101.js the string after 'myDeviceArray[0]' holds alll the data we need.
In status-102.js the string after 'var webData=' holds all the data we need.

Maybe you can use some of my code. I've a dedicated RPi configured for testing porposes so feel free to present some code for testing.

Sorry for my lousy coding but for the status-101.js variant I'm currently using the following code part:
```
     elif (brand == "Omnik2"):
       ip            = config.get(Inverter,'ip')
        Inv_sn        = config.get(Inverter,'Inv_sn')
        status = ""

        link = "http://"+ip+"/js/status.js"
        lines = urllib.urlopen(link).readlines()

        #Find line with serial(id) of device
        for line in lines:
            if Inv_sn in line:
                status = line

        # Remove parts of the string that are not needed
        status = status.replace("myDeviceArray[0]=","")
        status = status.replace(";","")
        status = status.replace("\"","")
        status = status.replace("\n","")

        # Convert to array
        statusList = status.split(',')

        S_InvID  = statusList[0]
        S_PAC1   = int(statusList[5])
        S_EToday = str(float(statusList[6])/100)
        S_ETotal = str(float(statusList[7])/10)
        S_HTotal = 0
```

And for status-102.js variant I've the following in place:
```
    elif (brand == "Omnik3"):
        ip            = config.get(Inverter,'ip')
        Inv_sn        = config.get(Inverter,'Inv_sn')
        status = ""

        link = "http://"+ip+"/js/status.js"
        lines = urllib.urlopen(link).readlines()

        #Find line with serial(id) of device
        for line in lines:
            if Inv_sn in line:
                status = line

        status2 = status.split(";")

        #Find line with serial(id) of device
        for line in status2:
            if Inv_sn in line:
                status3 = line

        # Remove parts of the string that are not needed
        status3 = status3.replace("var webData=","")
        status3 = status3.replace("\"","")
        status3 = status3.replace("\n","")

        # Convert to array
        statusList3 = status3.split(',')

        S_InvID  = statusList3[0]
        S_PAC1   = int(statusList3[5])
        S_EToday = str(float(statusList3[6])/100)
        S_ETotal = str(float(statusList3[7])/10)
        S_HTotal = 0
```

Below code prints the output in the correct format.
```
    if options.verbose:
    # For output to screen
        print "Date: ", S_Date
        print "ID: {0}".format(S_InvID)
        print "P Actual:{0:>7} W".format(S_PAC1)
        print "E Today: {0:>7} kWh".format(S_EToday)
        print "E Total: {0:>7} kWh".format(S_ETotal)
```

[status.js files.zip](https://github.com/jbouwh/omnikdatalogger/files/7463207/status.js.files.zip)
[image]
[image]

jbouwh, 2021-11-02:
Thnx I'll have a look.

jbouwh, 2021-11-03:
I saw this plugin (there are more forks): https://github.com/rubenOF/Domoticz-Omnik-Local-Web-Plugin/blob/master/plugin.py
Can't really tell if that will support both of your payload variants, have to find that out.

georgeavd, 2021-11-03:
No worry I've my old solution still in place 😉

jbouwh, 2021-11-04:
Just to make sure: No authentication is required to access the `/js/status.js` file?

jbouwh, 2021-11-04:
I have created a pre-release which adds http fallback to tcpclient. Could you test the code?

georgeavd, 2021-11-04:
Thanks for providing a code update on such a short notice.
The source code looks good but I will first be able to test these on saturday.

georgeavd, 2021-11-07:
I successfully tested the update of `cpclient.py` on the `status-101.js` variant but was not able to test it against the `status-102.js` veriant because port 8899 is providing data for this inverter.

But some strange behaviour was noticed after running `sudo omniklogger.py -d --interval 20`:
1. Data could only be read from the 101-variant when it was the only inverter being configured on `config.yaml` and there was a 30 second delay before the routine switched to reading `status,js`. After the first attempt there was no delay anymore and now every 20 secondes `status.js` was read without checking port 8899.
2. When all 3 inverters (-100, -101 and -102) are configured in `config.yaml` there is a timeout when waiting on port 8899 for data of the 101-inverter and no data is retrieved from the `js/status.js` file for this inverter.

It would be nice if you can provide an update with the option to configure in the config.yaml file if the tcpclient should read data from port 8899 or reading the data from http://<ip>/js/status.js
This would save the first timeout when the tcpclient routine is waiting for data from port 8899 which is not coming.
It will also provide the option of testing against the 102-variant of the inverter.

jbouwh, 2021-11-07:
I will add an option `http_only` and build a next beta.

jbouwh, 2021-11-07:
Created a new beta with the new option. It need to be set the the plant specific section. Only one measurement will be taken. Then the script will wait until dawn. Let me now if this works. I tested the code with both your status files.

jbouwh, 2021-11-08:
Additional the cache for http fallback is now stored per inverter (plant). This is in case no `http_only` flag is configured.

georgeavd, 2021-11-08:
Thanks for the good work and quick response.

The update works perfectly with for now the only remark that I found that the `http_only` flag is found to be mandatory.
That's not a big deal for me, but data could only retrieved from inverter-100 (running with a working port-8899 server) when in the `config.yaml` file `http_only: no` was added for this inverter.

Please check this otherwise this will impact all other users when they perform an update of the scripts. They will now end up with no data.

jbouwh, 2021-11-08:
> the only remark that I found that the `http_only` flag is found to be mandatory

I have found the cause. Will fix that before releasing.

closed 2021-11-08 by jbouwh