jbouwh/omnikdatalogger #53: Strange behaviour of consumed energy after inverter shuts down after sunset

joost01, 2021-12-06:
I've noticed some strange behaviour of consumed energy after inverter shuts down after sunset.

See the attached screenshot:
[image]

It is visible on the pvoutput.org website when you open the stats of one of the previous days. When you look at the outputs, in this case for 04-12, you see the last inverter update is at 04:35 pm. After that the used energy column changes from 6.637 to 12.720 kWh. It stays at this value until the end of the day.

I'm not sure yet if it is the omniklogger script or something in PVoutput, but another account which is provided by the dsmr entity doesn't show this behaviour

jbouwh, 2021-12-06:
After the inverter stops generating only consumed energy is published, may be this is causing the artifact. What I can do is keep pushing the last produced energy from cache.

joost01, 2021-12-07:
Well, in my case the problem occured for the first time when I updated the docker container. The running container was a little bit outdated (created 2020-10-02), but this one didn't have this behaviour. I've attached a screenshot.

[image]

jbouwh, 2021-12-07:
Created a new beta release that will continue to send the generated energy from cache.

jbouwh, 2021-12-07:
This behaviour  has not been there all the time, so hopefullly we get some better detail with this new version. If there still is a delta, then there is a possible problem in the calculation. Another theory is that pvoutput has changed it algorithm.

jbouwh, 2021-12-07:
As it looks to me, there is something wrong with the calculation. I still see the same symptoms.

jbouwh, 2021-12-07:
Checked again. I see the data that is published is correct. It is also publish to home_assistant as MQTT sensor, and I do not see the jump there.

jbouwh, 2021-12-09:
[image]
pvoutput seems to get correct data, so I cannot explain the behavior. The energy used seems to hang is some sort of way. But the pvoutput sensor (homeassistant) reports the correct energy_consumed. This is not what the graph tells.

jbouwh, 2021-12-09:
[image]

jbouwh, 2021-12-09:
This is while not sending generated energy. Some how it then uses the time stamp of the last generated energy data. All datapoints are raised together, but the total energy is correct.
[image]

jbouwh, 2021-12-10:
Somehow there is a duplicate upload, that uses the last solar timestamp. Will sort this out later.
```
2021-12-10 16:20:12.308665 INFO omnikdatalogger: pvoutput upload: {'d': '20211210', 't': '15:48', 'c1': 1, 'v3': '12930738.000', 'v4': '396.000', 'v1': '610.00', 'v2': '0', 'v6': 232.0}
2021-12-10 16:20:11.701746 INFO omnikdatalogger: Combining cached logging for plant 953  with DSRM data.
2021-12-10 16:20:11.698968 INFO omnikdatalogger: pvoutput upload: {'d': '20211210', 't': '16:20', 'c1': 1, 'v3': '12930738.000', 'v4': '396.000', 'v1': '610.00', 'v2': '0', 'v6': 232.0}
2021-12-10 16:20:11.104854 INFO omnikdatalogger: Net data update for plant 953 UTC 2021-12-10 15:20:08+00:00
2021-12-10 16:18:32.331497 INFO omnikdatalogger: pvoutput upload: {'d': '20211210', 't': '15:48', 'c1': 1, 'v3': '12930727.000', 'v4': '405.000', 'v1': '610.00', 'v2': '0', 'v6': 233.0}
2021-12-10 16:18:31.728129 INFO omnikdatalogger: Combining cached logging for plant 953  with DSRM data.
2021-12-10 16:18:31.725301 INFO omnikdatalogger: pvoutput upload: {'d': '20211210', 't': '16:18', 'c1': 1, 'v3': '12930727.000', 'v4': '405.000', 'v1': '610.00', 'v2': '0', 'v6': 233.0}
```

jbouwh, 2021-12-10:
I think I have solved the the issue now, and released a new beta. I'll let it run for some time to see if this is working correctly now.

jbouwh, 2021-12-11:
New release and fix docker version should solve the issue.

closed 2021-12-11 by jbouwh

joost01, 2021-12-13:
I've downloaded the latest image, if I notice any strange behaviour I'll let you know. Thank you again voor the quick response!

joost01, 2021-12-15:
For the last two days the data on PVOutput was as expected! Problem solved, ticket can be closed.