jbouwh/omnikdatalogger #62: DSMR via 3rd party device

ihulsbus, 2022-02-10:
Hi Jan, 

I currently have a HomeWizard Wi-Fi P1 meter connected to the central meter to export some power data. Now, this device sends it both to an remote system and exposes the data via a local, unprotected endpoint. I would like to see if it would be possible to integrate this data with the data from our solar panel inverters. Would it be possible to support such endpoints? 

I realize of course that it is impossible to support such devices all from the datalogger but maybe if users supply their own field mapping for the endpoint, assuming the data is exposed in JSON, it could work?

I'm currently looking at the following exported data: 
```
{
    "smr_version": 42,
    "meter_model": "Kaifa KAIFA-METER",
    "wifi_ssid": "<REDACTED>",
    "wifi_strength": 54,
    "total_power_import_t1_kwh": 19076.256,
    "total_power_import_t2_kwh": 16463.774,
    "total_power_export_t1_kwh": 3110.004,
    "total_power_export_t2_kwh": 7837.318,
    "active_power_w": 826,
    "active_power_l1_w": 219,
    "active_power_l2_w": 4,
    "active_power_l3_w": 603,
    "total_gas_m3": null,
    "gas_timestamp": null
}
```

Unfortunately I do not have this information for the gas part as it's not a "smart" gasmeter. 

If this cannot be supported, could I do the same data combining afterwards via my InfluxDB collections? Or does the logger do some magic internally to match the two sources together? 

Thanks,
Ian

ihulsbus, 2022-02-11:
Thanks. Looking at the output of the device, it seems to be a proprietary format, and not something publicly available, making integration difficult. 

I think I'll try to do the math in InfluxDB afterwards.

I got two additional problems I'm running into, one is the logger not able to authorize against influxDB with a token or basicAuth. The second one has to do with the scraping of an inverter. one can be scraped over 8899 (Inverter connected over WiFi) the other not (connected over Ethernet). Both are the same age inverter with a serial number that should be high enough (16XXXXXXXX and 19XXXXXXXX)

I'll create a separate issue for the first one, but I'm not sure about the inverter issue?

jbouwh, 2022-02-11:
If connecting over 8899 is possible for all inverters than that is ok. for Influx you need to enable v1 auth if you are using Influx v2

jbouwh, 2022-02-22:
> It is for one, but strangely enough not for the other. The main difference is connection medium and the serial number is of a higher number. Firmware on both is exactly the same. I'm still not sure what's going on there.
> 
> Saw the Influx v2 PR. I'll try it as soon as I have the chance. Thanks!

To reverse engineer the data  it is, to begin with a good thing to have data samples.
If you set the loglevel to debugging `tcpclient` will also log a base64 datadump if the data block is >= 99 bytes.