jbouwh/omnikdatalogger #85: tcpclient without inverter_sn?

goessensj, 2023-04-15:
Is it possible to connect to the inverter without a SN?
Since a software upgrade, my inverter is stripped from his serial number, the field is empty in the web interface, also external logging is broken.

I wanted to try to read the values on my local network via tcpclient.
The inverter is listening on port 8899.

I'm using the old serial number in the config file.
But i'm getting the following result:

INFO:omnik.datalogger:I 2023-04-15T19:48:43.310815 Initializing: Trying to reach the inverter for plant 123 over port 8899.
INFO:omnik.datalogger:I 2023-04-15T19:49:23.340375 Inverter is not reachable over port 8899, this is normal when it is dark. timed out

MarijnS95, 2023-04-15:
@goessensj The serial number is encoded in the TCP request package, but my 3000TL is totally happy with any random number I give it. It even replies with the correct serial number.

The error you shared _doesn't sound_ like related to a wrong S/N, but I am not familiar with the code. You could https://github.com/klaasnicolaas/python-omnikinverter in case it gives any more information/validation on the TCP packets.

goessensj, 2023-04-15:
The inverter is reachable.

```
pi@raspberrypi:~/OmnikDataLogger/Omnik-Data-Logger $ telnet <ip> 8899
Trying <ip>...
Connected to <ip>.
Escape character is '^]'.

```

I checked out the page you linked, my inverter is not listed, i have a 3000-TL3. Maybe that's the problem?

MarijnS95, 2023-04-16:
Unsure, it might also depend on the wifi/serial kit that is installed in the inverter.

Are you initiating the connection from the same machine? If so, and `telnet` connects successfully, perhaps it's just this Python package printing a generic error on "any" issue.

goessensj, 2023-04-17:
I run the telnet command on the same machine, except i run omnikdatalogger in docker.

jbouwh, 2023-10-24:
Not all inverters support a local tcp connection over 8899. In that case you could reroute the tcp packets and capture them. More info can be found on the WiKi pages. Closing this issue as it is stale.

closed 2023-10-24 by jbouwh

p4r4ll4x, 2023-10-25:
Why is the inverter listening on port 8899 if it wouldn't support such a connection?
Also, the script reports "Inverter is not reachable over port 8899", a manual connection is possible.

p4r4ll4x, 2023-10-25:
> Not all inverters support a local tcp connection over 8899. In that case you could reroute the tcp packets and capture them. More info can be found on the WiKi pages. Closing this issue as it is stale.

Why is the inverter listening on port 8899 if it wouldn't support such a connection?
Also, the script reports "Inverter is not reachable over port 8899", a manual connection is possible.

jbouwh, 2023-10-25:
Sorry, not the in Inverter but the wifi logger. There were some app features in the past that could use it. It never worked for me though.