jbouwh/omnikdatalogger #86: Datalogger in HACS throwing errors

rwrw01, 2023-10-24:
Installed HACS and appdeamon and then datalogger in HA 11. Appdeamon crashes when using dsmr but that is not a problem. The problem is that I can’t seem to start the app. Hereby the log from appdeamon:  

2023-10-24 21:47:21.823580 WARNING Error: ------------------------------------------------------------
2023-10-24 21:50:43.298566 WARNING Error: ------------------------------------------------------------
2023-10-24 21:50:43.298916 WARNING Error: Unexpected error loading module: /config/appdaemon/apps/omnikdatalogger/omniklogger.py:
2023-10-24 21:50:43.299195 WARNING Error: ------------------------------------------------------------
2023-10-24 21:50:43.305116 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/appdaemon/apps/omnikdatalogger/omniklogger.py", line 13, in <module>
    from omnik.datalogger import DataLogger
  File "/config/appdaemon/apps/omnikdatalogger/omnik/datalogger.py", line 20, in <module>
    from .dsmr import DSRM
  File "/config/appdaemon/apps/omnikdatalogger/omnik/dsmr/__init__.py", line 1, in <module>
    from dsmr_parser import obis_references
ModuleNotFoundError: No module named 'dsmr_parser'

2023-10-24 21:50:43.305558 WARNING Error: ------------------------------------------------------------

Some assistant ?

jbouwh, 2023-10-24:
Seems you are missing the import. You need to add them to the AppDaemon add-on config:

[image]

rwrw01, 2023-10-24:
Don’t need dsmr, further this package will crash appdeamon ?! I just pulled the apps.yaml from your reader.md ;)

jbouwh, 2023-10-24:
The imports are global. Did you try to add the imports. I have it running without issues.

rwrw01, 2023-10-24:
To fast. Just removed both omnikdatalogger and appdeamon and started over and got it working, strange but nice

closed 2023-10-24 by jbouwh