jbouwh/omnikdatalogger #96: This project is not available in HACS

Genesis1969, 2025-03-12:
Hello,

I am not sure this project is still maintained, if so i like to notify that this project could not found in HACS (HACS version 2.05, HA Core 2025.3.2)
However there is one other project however that shows with your build(s) [Elro Connects] when searching for dev name (jbouwh)

If this project is still supported, would you make it available with HACS (again) ?
(of course manual installation is possible i know.)

jbouwh, 2025-03-12:
Hi, the project is maintained (using it myself) and available as an AppDaemon integration. Not that the AppDaemon needs to be installed.

jbouwh, 2025-03-12:
[image]

closed 2025-06-02 by jbouwh