jbouwh/omnikdatalogger #102: Omnik Datalogger doesnot start with AddON Appdaemon versions higher than 0.16.7

florisdegroot, 2025-06-07:
I get this error while running Appdaemon version 0.17.4. This I see in the log off Appdaemon 4.x
The Datalogger still works with AppDaemon version 0.16.7, si I keep restoring this version.

2025-06-07 09:32:13.827785 INFO AppDaemon: New app config: omnikdatalogger
2025-06-07 09:32:14.534579 INFO AppDaemon: Starting apps: {'omnikdatalogger'}
2025-06-07 09:32:14.539837 WARNING AppDaemon: App 'omnikdatalogger' failed to start
2025-06-07 09:32:14.541975 ERROR Error: =====  Failed to start 'omnikdatalogger'  =================================
2025-06-07 09:32:14.542880 ERROR Error: AppStartFailure: App 'omnikdatalogger' failed to start
2025-06-07 09:32:14.546390 ERROR Error:   InitializationFail: initialize() method failed for app 'omnikdatalogger'
2025-06-07 09:32:14.548329 ERROR Error:     BadInitializeMethod: <class 'omniklogger.HA_OmnikDataLogger'> has a bad initialize method
2025-06-07 09:32:14.548840 ERROR Error:       HA_OmnikDataLogger.initialize(*args, **kwargs)
2025-06-07 09:32:14.549320 ERROR Error:       apps/omnikdatalogger/omniklogger.py
2025-06-07 09:32:14.550779 ERROR Error: ===========================================================================
2025-06-07 09:32:14.551289 INFO AppDaemon: App initialization complete
2025-06-07 09:32:57.693540 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-06-07 09:32:57.888698 INFO AppDaemon: Stopping apps from namespace 'default' because the plugin failed
2025-06-07 09:32:57.889991 INFO AppDaemon: Stopping apps: {'omnikdatalogger'}
2025-06-07 09:32:57.891325 INFO AppDaemon: Calling terminate() for 'omnikdatalogger'
2025-06-07 09:32:58.031422 INFO omnikdatalogger: Stopping Omnikdatalogger...
2025-06-07 09:32:58.033216 WARNING omnikdatalogger: ------------------------------------------------------------
2025-06-07 09:32:58.033658 WARNING omnikdatalogger: Unexpected error running terminate() for omnikdatalogger
2025-06-07 09:32:58.034129 WARNING omnikdatalogger: ------------------------------------------------------------
2025-06-07 09:32:58.038167 WARNING omnikdatalogger: Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/appdaemon/app_management.py", line 298, in terminate_app
    await utils.run_in_executor(self, term)
  File "/usr/lib/python3.12/site-packages/appdaemon/utils.py", line 559, in run_in_executor
    return await future
           ^^^^^^^^^^^^
  File "/usr/lib/python3.12/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/omnikdatalogger/omniklogger.py", line 162, in terminate
    self.rt.stop()
    ^^^^^^^

jbouwh, 2025-06-07:
Hi I know there is an issue with startup. There are some newer versions of omnikdatalogger published, but not sure that you have already seen those.

florisdegroot, 2025-06-07:
> To make sure you have the latest version from HACS, and not an old copy. Set app_dir to app_dir: /homeassistant/appdaemon/apps This is where HACS downloads the latest version. If you do not set the the config parameter, the config within the addon data will be used.

Beste Jan,

Bedankt dat je naar mijn Issue hebt gekeken.

 

Ik heb alle updates t/m v1.15.3 van de Omnik Datalogger geïnstalleerd voordat ik het Issue opende.

Ook ben ik geheel up-to-date met HA Core en draai met vs. 2025.5.3 en Supervisor 2025.03.3

IK heb ook het gehele systeem een keer geheel herstart maar het issue blijft komen.

 

Vanmiddag heb ik nagekeken (nadat ik het Issue had aangemaakt)  wat je zoals geupdate had in de Omnik files, zoals in de omniklogger.py waarin je b.v. regel 118 hebt gewijzigd:

van:

def initialize(self, *args, **kwargs):

Naar:

def initialize(self)

 

Ik meld dit omdat ik in de logfile, zie mijn Issue, het volgende zie staan:

2025-06-07 09:32:14.548840 ERROR Error: HA_OmnikDataLogger.initialize(*args, **kwargs)

 

En in Mijn geupdate file omniklogger.py zie ik staan:

def initialize(self):

 

Mijn Files zijn dus up-to-date echter het lijkt er op dat Appdaemon nog de oude versie opstart en dus de Error blijft geven.

 

Ik hoop dat jij een suggestie hebt van Hoe dit probleem op te lossen.

 

Onder Appdaemon vs. 16.7 werkt alles goed!

 

Ps. Ik heb je laatste update (kwam vanavond binnen) van Omnikdatalogger naar v1.15.6 nog niet geïnstalleerd en uitgeprobeerd. Dat zal ik morgen ochtend doen.

 

Ik hoor graag van je ,

Met vriendelijke groeten,

Floris de Groot

florisdegroot, 2025-06-08:
Als laatste het volgende want ik kom niet verder:

 

IK zak dat ik ook een directory \\homeassistant\addon_configs <file://homeassistant/addon_configs>  had.

Foto1 geeft de onderliggende Apps structuur aan en zie ik de Apps.yaml staan.

Foto2 geeft de daaronder liggende Omnikdatalogger structuur aan en daar staat de omnikdatlogger.py file die NIET geupdate is.

 

Ik hoop dat ik hiermee meer duidelijk geef wat er in mijn systeem staat en kun jij met een oplossing komen want kennelijk komen bestanden dubbel voor en worden de verkeerde bestanden geupdate en opgestart?

 

Met vriendelijke groeten,

Floris de Groot

jbouwh, 2025-06-08:
The appdaemon location for appdaemon `apps` automation files from HACS defaults to `/homeassistant/appdaemon/apps` (seen from the appdeamin perspective).
By default `appdaemon` reads apps from `/addons_config/xxxxxxxx_appdaemon/apps` instead. Previous installs did not use the `addons_config`, so these files were copied once. To be able to maintain apps via HACS, you need to change the `app_dir` option under the `appdaemon` mapping in `appdaemon.yaml`. This file is in `/addons_config/xxxxxxxx_appdaemon/`.

My config file looks like:

```yaml
---
appdaemon:
  latitude: 52.379189
  longitude: 4.899431
  elevation: 2
  time_zone: Europe/Amsterdam
  app_dir: /homeassistant/appdaemon/apps
  plugins:
    HASS:
      type: hass
      token: !env_var SUPERVISOR_TOKEN
http:
  url: http://0.0.0.0:5050
admin:
api:
hadashboard:
```

florisdegroot, 2025-06-08:
> @florisdegroot <https://github.com/florisdegroot>  Can you contact me at Discord on @jbouwh <https://github.com/jbouwh>  ?

I will call you at about 14:30.

closed 2025-06-08 by florisdegroot

Floriszz, 2026-10-09:
> 10\. Issue has been solved by update in file \homeassistant\addon_configs\a0d7b954_appdaemon\appdaemon.yaml.
>      Below line has been added under the line   time_zone: Europe/Amsterdam
>      app_dir: /homeassistant//appdaemon/apps
>      This above directory contains the updated version of omnikdatalogger.

I don't think you need(should use) the 'double slashes' between /homeassistant and appdaemon ? Or do you?
I'm also fighting through this confusing amount of documentations with different file paths...😥